adafruit/Adafruit_CircuitPython_Crickit #14: MemoryError

skyward-xn, 2019-02-17:
Hi!

I use Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit CircuitPlayground Express with samd21g18

The code `from adafruit_crickit import crickit` always throws

```
Traceback (most recent call last):
  File "main.py", line 1, in <module>
  File "adafruit_crickit.py", line 66, in <module>
MemoryError: memory allocation failed, allocating 152 bytes
```

Anybody has the same problem?

ladyada, 2019-02-17:
make sure you don't have the crickit or seesaw library in `lib` its built in already!

jerryneedell, 2019-02-17:
I think you will need to install the build explicitly for the circitplaygound_express_crickit.
I looks like you have the standard circlutplayground_express build installed.
adafruit-circuitpython-circuitplayground_express_crickit-3.1.2.uf2
it has the cricket library "builtin"
more information here: https://learn.adafruit.com/adafruit-crickit-creative-robotic-interactive-construction-kit/circuitpython-code

skyward-xn, 2019-02-18:
Thank you a lot for responses! I'll try it out today and close the issue if it's OK. I actually have a lib folder and may have used a wrong build :)

skyward-xn, 2019-02-18:
Tried it out and it works! Thank you a lot.

closed 2019-02-18 by skyward-xn

ladyada, 2019-02-18:
@jedgarpark @TheKitty this error comes up a bunch due to not using the crickit build and/or having stuff in `lib` - please add to the guide and ill QA