adafruit/Adafruit_CircuitPython_Crickit #15: Stepper broken due to change in Adafruit_CircuitPython_Motor

caternuson, 2019-03-13:
Started here:
https://forums.adafruit.com/viewtopic.php?f=8&t=148916
and verified. This PR:
https://github.com/adafruit/Adafruit_CircuitPython_Motor/pull/23/files
changed the way the `frequency` value is checked. It now throws an Error instead of changing it.

The stepper setup here needs to set `frequency` on its `PWMOut` objects. This might be made easier if `frequency` could be passed in to PWMOut constructor:  
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/master/adafruit_seesaw/pwmout.py#L29

ladyada, 2019-03-14:
@siddacious heya can you check this and revert ? ive been swamped and unable to test

siddacious, 2019-03-14:
Yup, I'll take a look.

siddacious, 2019-03-14:
OOf. My bad. :|

ladyada, 2019-03-19:
hiya this should be all fixed now!

closed 2019-03-19 by ladyada