adafruit/Adafruit_CircuitPython_Crickit #19: Reusing crickit I2C instance

dhalbert, 2019-08-07:
`crickit` creates its own `I2C` instance, instead of using `board.I2C()`. It was written before `board.I2C()` existed.

It should use `board.I2C()` if possible (maybe we can assume it exists now in all supported versions of CircuitPython). This will allow external I2C devices to be used easily.

Alternatively, or in addition, we could have a property that returns `crickit.seesaw.i2c_device` and/or the underlying `i2c'.

tagging @caternuson, since this has come up in support issues.

caternuson, 2019-08-07:
It's currently created conditionally:
```python
if "SCL" in dir(board):
    crickit = Crickit(Seesaw(busio.I2C(board.SCL, board.SDA))) # pylint: disable=invalid-name
```
Could we just change that up a bit:
```python
if "I2C" in dir(board):
    crickit = Crickit(Seesaw(board.I2C())) # pylint: disable=invalid-name
```

dhalbert, 2019-08-07:
I looked, and `board.I2C()` was added in 3.x. So that seems fine: we don't need to support 2.x any more. We could still add a `.i2c` property but it's not vital.

caternuson, 2019-08-07:
Older versions could still get there, right? Would just need to manually setup the crickit instance after importing.

And the `.i2c` property idea might be confusing. There'd be two different `foo_i2c_sensor()` init's depending on if crickit is being used or not.

dhalbert, 2019-08-07:
> Older versions could still get there, right? Would just need to manually setup the crickit instance after importing.

Right.

PR #20 (2019-08-07): Switch to using I2C singleton
Fixes #19 

```python
Adafruit CircuitPython 4.0.2 on 2019-06-27; Adafruit Feather M4 Express with samd51j19
>>> from adafruit_crickit import crickit
>>> crickit.touch_1.value
False
>>> crickit.touch_1.value
True
>>> import board, busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SCL in use
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> i2c.scan()
[19, 73]
>>> i2c.unlock()
>>> crickit.touch_1.value
False
>>>
```
closed 2019-08-07 by dhalbert