adafruit/Adafruit_CircuitPython_Crickit #22: Add support for on-board Seesaw NeoPixel

CedarGroveStudios, 2019-12-05:
Would it be possible to expose the Crickit's on-board NeoPixel (Seesaw pin 27) from the _Crickit_ class? This is what it takes to activate and use it with CircuitPython 4.1.0:

```
from adafruit_crickit import crickit
crickit.init_neopixel(0)
ss = crickit.seesaw

# set up on-board Seesaw NeoPixel
from adafruit_seesaw.neopixel import NeoPixel
sspixel = NeoPixel(ss, 27, 1, brightness=0.01)

sspixel.fill([128, 0, 0])
```

ladyada, 2019-12-05:
sure if you like to submit a PR we'll take a look

CedarGroveStudios, 2019-12-05:
Thanks. I looked at the code and it's a bit beyond my skills -- but I'll give it a shot.

CedarGroveStudios, 2019-12-05:
By jove, I think I've got it! Here's the UI:
```
from adafruit_crickit import crickit
crickit.init_onboard_pixel(brightness=0.01)
crickit.onboard_pixel[0] = ((255, 24, 255))
```

The _init_onboard_pixel()_ argument structure is similar to _init_neopixel()_:
```
def init_neopixel(self, n, *, bpp=3, brightness=1.0, auto_write=True, pixel_order=None):

def init_onboard_pixel(self, *, n=1, bpp=3, brightness=1.0, auto_write=True, pixel_order=None):
```
I'll submit a PR in a few minutes.

closed 2019-12-05 by CedarGroveStudios