adafruit/Adafruit_CircuitPython_Crickit #30: Add DC Motor PWM Frequency Attribute

CedarGroveStudios, 2021-01-29:
The ability to adjust the PWM frequency of the Crickit's DC motor outputs would allow optimizing a given motor's torque and speed characteristics. This is particularly useful for increasing the motor's low-speed torque and dramatically improves the speed control dynamic range for most brushed DC motors.

See the related MotorKit change:  [https://github.com/adafruit/Adafruit_CircuitPython_MotorKit/pull/37](url)

tannewt, 2021-02-01:
I think your usage is advanced enough that it'd be better for you to use MotorKit directly in this case. At some point libraries are abstracting away details and lose API granularity.

CedarGroveStudios, 2021-02-01:
Anyone attaching a TT motor or other DC motor using just the Crickit library will not be able to take advantage of the motor's full torque range, particularly at low RPMs.

But I'm willing to to back to the drawing board and find a way to continue to use other Crickit functionality along with MotorKit. Was hoping for a simple solution for the upcoming learning guide.

tannewt, 2021-02-02:
This is a consequence of the module instantiating the crickit: https://github.com/adafruit/Adafruit_CircuitPython_Crickit/blob/master/adafruit_crickit.py#L402

Really easy libraries can't always do everything. The lower level libraries should be nearly as accessible but more feature-ful.

CedarGroveStudios, 2021-02-02:
Understood. Appreciate your insight.

I'll take this off-line and will use it as a personal educational opportunity. Thanks!

CedarGroveStudios, 2021-02-15:
Found a workaround to set the PWM frequency of an individual Crickit Seesaw pin that maintains the UI for motor throttle control. This will be included in the learning guide. Tested on the Crickit FeatherWing and Circuit Playground Express Crickit.

```python
from adafruit_crickit import crickit

PWM_FREQ = 25  # Custom PWM frequency; Crickit default is 50Hz

motor_1 = crickit.dc_motor_1
motor_2 = crickit.dc_motor_2

ss = crickit.seesaw  # To access Seesaw motor pins
ss.set_pwm_freq(22, PWM_FREQ)  # Set motor1A pin to custom PWM frequency
ss.set_pwm_freq(23, PWM_FREQ)  # Set motor1B pin to custom PWM frequency
ss.set_pwm_freq(19, PWM_FREQ)  # Set motor2A pin to custom PWM frequency
ss.set_pwm_freq(18, PWM_FREQ)  # Set motor2B pin to custom PWM frequency

motor_1.throttle = 0  # Stop motor_1
motor_2.throttle = 0  # Stop motor_2
```

Was happy to see that the default PWM frequency for the Crickit's DC motors is 50Hz unlike MotorKit's 1600Hz default. The lower frequency works nicely with TT motors.

closed 2021-02-15 by CedarGroveStudios

tannewt, 2021-02-16:
Perfect @CedarGroveStudios!