adafruit/Adafruit_CircuitPython_Crickit #31: Missing Type Annotations

FoamyGuy, 2021-09-27:
There are missing type annotations for some functions in this library.

The `typing` module does not exist on CircuitPython devices so the import needs to be wrapped in try/except to catch the error for missing import. There is an example of how that is done here:

```python
try:
    from typing import List, Tuple
except ImportError:
    pass
```

Once imported the typing annotations for the argument type(s), and return type(s) can be added to the function signature. Here is an example of a function that has had this done already:

```python
def wrap_text_to_pixels(
    string: str, max_width: int, font=None, indent0: str = "", indent1: str = ""
) -> List[str]:
```
If you are new to Git or Github we have a guide about contributing to our projects here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

There is also a guide that covers our CI utilities and how to run them locally to ensure they will pass in Github Actions here: https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/check-your-code In particular the pages: `Sharing docs on ReadTheDocs` and `Check your code with pre-commit` contain the tools to install and commands to run locally to run the checks.

If you are attempting to resolve this issue and need help, you can post a comment on this issue and tag both @foamyguy and @kattni or reach out to us on Discord: https://adafru.it/discord in the `#circuitpython-dev` channel.

The following locations are reported by mypy to be missing type annotations:

- [ ] adafruit_crickit.py:85
- [ ] adafruit_crickit.py:145
- [ ] adafruit_crickit.py:210
- [ ] adafruit_crickit.py:244
- [ ] adafruit_crickit.py:290
- [ ] adafruit_crickit.py:318
- [ ] adafruit_crickit.py:334

tcfranks, 2022-07-08:
I was going to work on this, but feel 'in over my head', perhaps.  For example:   A return type of Any seems like it is an over simplification of theses return types, but since they could return an object | None, it it appropriate?  

 @property
    def feather_drive_stepper_motor(self):
        """``adafruit_motor.motor.StepperMotor`` object on Drive terminals on Crickit FeatherWing"""
        return self._motor(tuple(reversed(_DRIVE_STEPPER)), StepperMotor)

    def _motor(self, terminals, motor_class):
        device = self._devices.get(terminals, None)
        if not isinstance(device, motor_class):
            device = motor_class(
                *(PWMOut(self._seesaw, terminal) for terminal in terminals)
            )
            self._devices[terminals] = device
        return device

FoamyGuy, 2022-07-09:
There is `Optional` keyword for when it can return a thing or `None` i.e.:

```
from typing import Optional

def some_function(arg1: int) -> Optional[int]:
    if arg1 > 42:
        return 178
    return None
```

It can be done similarly for object types instead of primatives like `int`  here is another example with an object type instead:

```
import board
import microcontroller
def get_neopixel_pin() -> Optional[microcontroller.Pin]:
    if board.hasattr("NEOPIXEL"):
        return board.NEOPIXEL
    return None
````

It should work similarly for `StepperMotor` or any classes more specific to this library. If you're still stuck a bit though let me know and I can take a closer look at the specific chunk of code in this library and try to offer guidance on type.

tcfranks, 2022-07-09:
Thanks!   That means I have a couple of things to fix here and on the circuitplayground code base!

patenaud, 2023-04-26:
Pycon2023 - Will give this a shot.

closed 2023-06-05 by FoamyGuy