adafruit/Adafruit_CircuitPython_Crickit #35: Using crickit.onboard_pixel raises an import ModuleNotFoundError for adafruit_pixelbuf

marwano, 2022-08-06:
I have setup my Python environment on a Raspberry Pi using the [Adafruit CRICKIT HAT for Raspberry Pi Linux Computers](https://learn.adafruit.com/adafruit-crickit-hat-for-raspberry-pi-linux-computers) guide. The below code used to work, but when I recreate the environment with the latest libraries it throws an import error because of a missing dependency in the installation.

I would create a fresh virtual environment and install the crickit libraries and dependencies with:

`pip3 install adafruit-circuitpython-crickit`

I then run the following code in the REPL to set the onboard NeoPixel to the color red.
```
from adafruit_crickit import crickit
crickit.onboard_pixel.fill([255, 0, 0])
```
This is the error I get
```
>>> crickit.onboard_pixel.fill([255, 0, 0])  # red
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/robo/pyenv/lib/python3.9/site-packages/adafruit_crickit.py", line 376, in onboard_pixel
    from adafruit_seesaw.neopixel import (  # pylint: disable=import-outside-toplevel
  File "/home/robo/pyenv/lib/python3.9/site-packages/adafruit_seesaw/neopixel.py", line 12, in <module>
    from adafruit_pixelbuf import PixelBuf
ModuleNotFoundError: No module named 'adafruit_pixelbuf'
>>> crickit.onboard_pixel.fill(0xFF0000)     # red
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/robo/pyenv/lib/python3.9/site-packages/adafruit_crickit.py", line 376, in onboard_pixel
    from adafruit_seesaw.neopixel import (  # pylint: disable=import-outside-toplevel
  File "/home/robo/pyenv/lib/python3.9/site-packages/adafruit_seesaw/neopixel.py", line 12, in <module>
    from adafruit_pixelbuf import PixelBuf
ModuleNotFoundError: No module named 'adafruit_pixelbuf'
```
After looking around at recent changes in the Adafruit libraries I found this change a few weeks ago
[rework neopixel to work with PixelBuf ](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/commit/a8b3958a9f28d4da24fe7b59f07aa3b2c67f4d56) that now needs the pixelbuf library to work. Then once I install the missing pixelbuf library with the below command, everything works fine like before.

`pip3 install adafruit-circuitpython-pixelbuf`

So I'm just raising it here to see maybe you want to add the adafruit-circuitpython-pixelbuf as a required dependency so that it gets automatically installed with adafruit-circuitpython-crickit library. Thanks.

tekktrik, 2022-09-15:
We recently updated our installation infrastructure, so it looks like some things didn't get ported correctly.  The fix is exactly what you mentioned, thanks for bringing it up!

tekktrik, 2022-09-15:
Actually, I remember there being a similar issue, but it was actually downstream.  The dependencies should be fixed now, but I'll confirm.

tekktrik, 2022-09-15:
Yup!  Fixing the original issue downstream fixed it here!  Thanks again!

closed 2022-09-15 by tekktrik