adafruit/Adafruit_CircuitPython_Crickit #39: Evaluating crickit in repl causes dc_motor_1 only to spin.

tyeth, 2024-09-11:
I'm playing with a 2wheel dc motor rover project.

Pressing tab in the REPL to evaluate available things off of `crickit` causes dc_motor_1 to fire but not dc_motor_2. 

Why is that, maybe while evaluating it's initialising stepper which causes stepper "poles" to be setup on the pins for dc_motor_1 that would be opposite to dc_motor_2's poles?

Turning off and on the motors individually works as expected.

I guess this is both a question to ask why it happens, and secondly a feature request to ask that it doesn't

ladyada, 2024-09-11:
you could run the cricket.py code and add pauses/prints to figure out where exactly its happening

tyeth, 2024-09-11:
So it turns out this is a "feature" / wish-list item in micropython https://github.com/micropython/micropython/issues/4546
Thanks to @jepler for pointing it out.

The one option appears to be providing a __dir__ function which is what tab complete also uses:
```
def __dir__(self):
        return []
```